VernierST/arduino #2: Thermistor Reading Discrepency

CorpusCallosum, 2015-03-03:
Hi, I'm using the temperature probes with the AnalogAutoID example Arduino sketch, but I'm getting about 10 degrees temperature discrepancy between analog port 1 and analog port 2, this happens even if I swap the sensors. Analog port 1 always reads about 10 degrees higher in temp than analog port 2. What might be causing this discrepancy? Thanks.

CorpusCallosum, 2015-03-03:
I was able to solve the problem by using the Thermistor example instead. There seems to be an issue with how the second analog value is being mapped with the AnalogAutoID example, but I couldn't find the root of the problem.

dvernier, 2015-03-04:
Hello,

At first, I was confused by your question because I assumed (incorrectly, I think) that you were using the sketch AnalogAutoIDTwoLineDisplay, This sketch is what we have been using mostly around here and it only works with one analog sensor.

I now see that you were using the simple sketch from last year, which does two analog sensors. I verified the problem and I will try to resolve it soon.

I the mean time, unless you need the autoID feature, you can do as you mentioned and use VernierThermistor or code based on that.

Dave

dvernier, 2015-04-02:
Hello,

I fixed the VernierAnalogAutoID example. We will be posting the fixed version soon. Sorry it took so long. The problem was that you need to turn off I2C communication at the end of the setup section using 
PinMode A4, INPUT;
PinMode A5, INPUT;

Dave Vernier

Smythics, 2015-04-02:
This has been posted!

closed 2015-04-02 by Smythics