VernierST/arduino #3: Sketch to display and graph readings to LabVIEW from 2 analog sensors

YukiSakuma, 2016-02-15:
Hello, I have 4 analog sensors namely, stainless steel temp, DO probe, pH probe and Ammonium Ion-Selective Electrode. I have two sparkfun arduino board. Now since the vernier interface shield has only two analog inputs I have two of them in order to connect all 4 sensors. So what I need is the sketch and possibly the VI to display each table and graph readings from these sensors. Any help is appreciated thanks.

YukiSakuma, 2016-02-16:
I was able to use the thermistor.ino code for stainless steel temp probe and display the readings to LabVIEW using this VI:http://forums.ni.com/ni/attachments/ni/170/741210/2/read_serie_monitor.vi
This was then I realized that I need the "actual" code for each probes just like the thermistor.ino for the stainless temp probe. I used the AnalogAutoID to identify the sensors but I need the "actual" code because I will use the readings of each probes to control a solenoid valve for example. I am using the 4 sensors in a giant aquarium with Tilapia fish in it.

Smythics, 2016-02-17:
Hi Yuki, 
Sounds like a great project!
I need to warn you about a couple of things. You are likely to run into some interference between pH and the Ammonium ISE when testing the same body of water. You might be able to get away from this by having them at opposite ends of the tank, but more likely you will need to pull a sample to test one or the other. The DO may also experience some interaction with the ISE but it is not as sensitive. Since you are already using two Arduino's that may work to your benefit. Put the ISE and the temperature on one end, pH and DO on the other. No guarantees, but...
If you want a custom code for the particular sensors you are using look at the sketch named VernierAnalogSensor.ino and modify it for the pH and DO sensors which have a linear calibration. You will need to look up the slope and intercept values in the sensor booklets for these sensors: http://www.vernier.com/support/manuals/
The ISE is a different animal altogether. The ISE voltage output is not linear with respect to the ion concentration, but rather is modeled by the Nernst Equation. In order to determine the concentration the mV response must be entered into the Nernst equation:
E = Eo + m(ln C)
The trick with the ISEs and the Nernst equation is that the values Eo and m change as the membrane slowly deteriorates. Every 2 weeks or so the membrane has changed enough to require a new calibration.
To determine the values of Eo and m, use 2 known concentration solutions. Record the voltage output of the ISE in each known solution and use then to solve for Eo and m. You can only expect these values to remain valid for a short period of time (a week or two) and then you will need to do this again. Be sure to update these values in your Arduino sketch. I have a sample sketch for the ISE that I will provide outside of this forum.

Smythics, 2016-02-17:
[VernierBNC (2).zip](https://github.com/VernierSoftwareTechnology/arduino/files/133523/VernierBNC.2.zip)
Here is the code for the ISE. Take a look and let me know if you have questions.

closed 2016-02-17 by Smythics

YukiSakuma, 2016-02-18:
Thanks for the detailed response but I think I found a way to use the AnalogAutoID to output a specific channel, like the statement of the for loop for reading Channels 1 to 2, I removed them and replaced something like
int Channel=1;
int Channel2=2; then

SensorReading[Channel]= Intercept[Channel] + Voltage[Channel] * Slope[Channel];// for analog channel 1

SensorReading[Channel2]= Intercept[Channel] + Voltage[Channel] * Slope[Channel];// for analog channel 2
then I specifically plug my temperature probe to channel 1 and dissolve oxygen in channel 2 then I serial print and I was able to output readings from multiple sensors into LabVIEW and graph its reading versus time. 

**Is re-calibration not needed on all of my four sensors when using AnalogAutoID?**

I have a follow up question can I use the Digital 1 and 2 of the interface shield as an output to control like opening a solenoid valve without using your DCU product any guide on how to do that?

My plan is let's say

if ( SensorReading[Channel]>threshold) //temp probe, threshold=30

digitalWrite(Digitalpin1, HIGH); // output 5 volts to a relay driver that would activate the solenoid valve

YukiSakuma, 2016-02-18:
Ok I think I might have fixed the reading for channel 2 this is what I did let me know if this is correct

```
void loop()
 {
    int Count[3]; //reading from 0 to 5 volt input
    int AltCount[3] ;//reading from -10 to +10V input
    float Voltage[3];
    float SensorReading[3];
    float Time;
    digitalWrite(led, HIGH); //turn on LED
//    Serial.print(ReadingNumber/1000.0*TimeBetweenReadings); 
    for (int Channel=1;Channel<3;Channel++) //2 was 3
      {
//        Serial.print("\t"); //tab character
        if (Name[Channel]=="Voltage +/- 10V")
          {
             AltCount[1] = analogRead(A1); //read both +/- 110 volt lines
             AltCount[2] = analogRead(A3);
              // convert from count to raw voltage if using 10 volt range:
              Voltage[Channel] = AltCount[Channel] / 1024.0 * 5.0 ;          
           }
          else
            {        
              Count[1] = analogRead(A0); //read both analog lines
              Count[2] = analogRead(A2);
              // convert from count to raw voltage on 0 to 5 range:
              Voltage[Channel] = Count[Channel] / 1024.0 * 5.0; 
            }
           SensorReading[Channel]= Intercept[Channel] + Voltage[Channel] * Slope[Channel];
     //special calibration for thermistor temperture probe:
     if (SensorNumber[Channel]==10) SensorReading[Channel]=Thermistor(Count[Channel]);
   **   if (Channel==1)
      {Serial.print("a");
       Serial.print(SensorReading[Channel]);  
       Serial.println(" "); 
       delay(50);
      }
      else if (Channel==2)
      {Serial.print("b");
       Serial.print(SensorReading[Channel]);  
       Serial.println(" "); 
       delay(50);
      }**
      else 
      {}
        
```
I referenced my modification in order to view the readings from multiple sensors to LabVIEW from this link:http://forum.arduino.cc/index.php?topic=83008.msg624557#msg624557

Smythics, 2016-03-08:
Sorry for the delay in response. Was out for a bit and still catching up.

DO, pH and ammonium ion will likely require routine calibration. Please note that these sensors are designed for education purposes and are not intended for prolonged use. 

Your project is pretty unique. Can you restate any remaining questions for me.

YukiSakuma, 2016-03-08:
Yes the project is all about creating a balanced environment for Tilapia growth (starting from fry or fingerling) by maintaining safe levels of pH, temp, DO and Ammonium (supposedly Ammonia). I do plan to have 3 trials where each trial is approximately 1 month each. I plan to use these sensors for long term monitoring (i.e. 1 month each) though...
Will try to formulate questions later thanks.

YukiSakuma, 2016-09-06:
Hello it's been a while the project got a little delayed but let's not bother with that. 
I now have the Tilapia fingerlings in the aquarium tank but before I could start the long term experiment I need to fix some concerns and issues.

My concern is about the Ammonium ISE calibration, I am using the Vernier BNC sketch you provided and it's very helpful but my concern is every time I calibrate in either solution 1 mg/L or 100 mg/L the electrode voltage reading increases over time. For example for the first minute the reading is 240mV but then after 5 minutes the reading is 260mV. What I do  is I get the highest electrode reading voltage after 5 minutes on both solutions to calibrate. Is this correct? I plan to recalibrate every a week instead of the mentioned 2 weeks in the comments

Next issues.
My first issue is also about the Ammonium ISE probe. I said in my concern about how to calibrate but just this day I am trying to recalibrate it but I am getting a constant electrode voltage reading (i.e. 490mV) in both solutions or in water. Any idea how to diagnose and fix this?

My second issue is about the DO and pH interference. Let's recap I have two Arduinos connected to a one computer;
Arduino 1 - Temp and Ammonium
Arduino 2 - pH and DO
I am getting an interference on pH and DO even when I put the probes on each end of the aquarium tank, the length of the aquarium tank is 54 inches. 
So I tried putting the DO probe in my Aquarium Tank filter in which water pours down from a PVC pipe, the interference lessened a little but the point is I am still getting an interference.
I see in this page also (http://www.vernier.com/til/files/638/sensor_interference_tables.pdf) that DO is bad with ISE in this case the Ammonium ISE.

dvernier, 2016-09-06:
> files/638/sensor_interference_tables.pdf) that DO is bad with ISE in this

Hello,

I will respond to your email, section by section. First, I want to say, as
we say all over our web site and on all of our documentation:  Our products
are designed for use in education. They are not meant for use in a
commercial fish farming situation.

>>My concern is about the Ammonium ISE calibration, I am using the Vernier
BNC sketch you provided and it's very helpful but my concern is every time
I calibrate in either solution 1 mg/L or 100 mg/L the electrode voltage
reading increases over time. For example for the first minute the reading
is 240mV but then after 5 minutes the reading is 260mV. What I do is I get
the highest electrode reading voltage after 5 minutes on both solutions to
calibrate. Is this correct?

I talked with a chemist who knows this sensor well and she said everything
you said makes sense. You have to be patient during the calibration
process. She says waiting 5 minutes at each calibration point is probably
about right.

>>My first issue is also about the Ammonium ISE probe. I said in my concern
about how to calibrate but just this day I am trying to recalibrate it but
I am getting a constant electrode voltage reading (i.e. 490mV) in both
solutions or in water. Any idea how to diagnose and fix this?

Obviously something is seriously wrong if the voltage reading is not
changing in those very different solutions. If the voltage reading is not
changing, I would double check all connections. Also, note that ISE
membranes do not last long. I read your email to the chemist and she
thought it might be that the membrane needs replacing. See page 6 of:

http://www.vernier.com/files/manuals/nh4-bta.pdf

>>My second issue is about the DO and pH interference. Let's recap I have
two Arduinos connected to a one computer;
Arduino 1 - Temp and Ammonium
Arduino 2 - pH and DO
I am getting an interference on pH and DO even when I put the probes on
each end of the aquarium tank, the length of the aquarium tank is 54
inches.
So I tried putting the DO probe in my Aquarium Tank filter in which water
pours down from a PVC pipe, the interference lessened a little but the
point is I am still getting an interference.
I see in this page also (http://www.vernier.com/til/fi
les/638/sensor_interference_tables.pdf) that DO is bad with ISE in this
case the Ammonium ISE.

There will always be interference using these two probes in one tank of
water. The issue involves ground loops. The water acts as a conductor.

Dave Vernier

—

David L. Vernier
Founder and Co-President
Vernier Software & Technology
13979 SW Millikan Way
Beaverton, OR  97005
503-277-2299
1-888-Vernier (1-888-837-6437)
dvernier@vernier.com
www.vernier.com

YukiSakuma, 2016-09-08:
Wow! It's an honor to have our question answered by you Sir Vernier! Thank you for helping me with my inquiries! I tried working with the information you gave me and I have some more questions.

My Dissolved Oxygen Probe (http://www.vernier.com/products/sensors/dissolved-oxygen-probes/do-bta/) interferes with my pH sensor even when trying to place one sensor in the filter. I have found this sensor;  DO Optical (http://www.vernier.com/products/sensors/dissolved-oxygen-probes/odo-bta/) does not interfere. How differently does the Optical Probe work?

I tried to fix my Ammonium ISE. I unscrewed the membrane, carefully wiped the electric contacts, put it back in, and soaked the Sensor in High-Standard solution (100 mg/L) for 30 minutes. It functioned again, but reverted to to displaying on 490mV. I haven't had the probe for more than 6 months, but the manual says that it is guarantees to last at least a year. Is it still possible to replace it? I carefully followed the instructions for testing, usage, and storage.

I truly appreciate you help in answering my earlier questions. The information is really helping me since I'm learning in the way.

dvernier, 2016-09-08:
> products/sensors/dissolved-oxygen-probes/do-bta/) interferes with my pH
> products/sensors/dissolved-oxygen-probes/odo-bta/) does not interfere.

Hello,

My Dissolved Oxygen Probe (http://www.vernier.com/
products/sensors/dissolved-oxygen-probes/do-bta/) interferes with my pH
sensor even when trying to place one sensor in the filter. I have found
this sensor; DO Optical (http://www.vernier.com/products/sensors/dissolved-
oxygen-probes/odo-bta/) does not interfere. How differently does the
Optical Probe work?

If you look at the two manuals (which you can download from our web site),
you will see that they are totally different. The Optical DO sensor is much
more expensive, but does not interact electrically with the other
water-quality sensors, like the other one does.

I tried to fix my Ammonium ISE. I unscrewed the membrane, carefully wiped
the electric contacts, put it back in, and soaked the Sensor in
High-Standard solution (100 mg/L) for 30 minutes. It functioned again, but
reverted to to displaying on 490mV.

I talked with the  chemists about this and they said that the key thing is
to look at the voltages out when the probe is in the two standards. They
should be 2.1 volts and 1.3 volts. Of course, these tests should be done
with no other sensors interfering and following all the procedures in the
user manual.

>>I haven't had the probe for more than 6 months, but the manual says that
it is guarantees to last at least a year. Is it still possible to replace
it? I carefully followed the instructions for testing, usage, and storage.

First, I have some questions:

- Do you have Vernier interface (LabPro, LabQuest, LabQuest 2, Go!Link) to
try this DO probe with?

- Have you done the tests with the two standards? What voltages do you get?
Are you saying that the voltage is 0.490 for each?

- Are you sure you have done all of this calibrating and testing with no
other sensor in water at the same time?

Let me know the answers. I will get out a DO probe and an Arduino here and
do some tests and compare results.

I do not want to encourage you to spend more money on an ODO probe. I am
not sure you really should be using our sensors in this project.

Dave Vernier

David L. Vernier
Founder and Co-President
Vernier Software & Technology
13979 SW Millikan Way
Beaverton, OR  97005
503-277-2299
1-888-Vernier (1-888-837-6437)
dvernier@vernier.com
www.vernier.com

YukiSakuma, 2016-09-09:
Greetings!

Regarding the Ammonium ISE, I tested them in an isolated environment.

```
"- Do you have Vernier interface (LabPro, LabQuest, LabQuest 2, Go!Link) to
try this DO probe with?

- Have you done the tests with the two standards? What voltages do you get?
Are you saying that the voltage is 0.490 for each?

- Are you sure you have done all of this calibrating and testing with no
other sensor in water at the same time?

Let me know the answers. I will get out a DO probe and an Arduino here and
do some tests and compare results."

```
*these all pertain to the Ammonium ISE

The only interface I'm using is a SparkFun Red Arduino Uno Board connected to a Sparkfun Vernier-Arduino Shield since Arduino can comminicate with LabView (my database).

The Raw voltage is 3-4V (regardless of solution) while the converted voltage (ElectrodeReading = 137.55*Raw Voltage -0.1682) sits at 490 mV.

Yes, I calibrated the ISE using only the two standard solutions and no other sensors even connected to the shield.

I also have one more question but for the pH sensor this time. When I measured the pH (already calibrated using the buffer solutions) of my 280-gallon fish tank, the final reading fluctates from 9-10. However, when I immediately move the sensor to a cup/glass of the same aqarium water, the final reading is around 7.2-7.4. This test was done with no other sensors in the water at the time. Does the volume of water have an effect on the output of the pH sensor?

I truly appreciate your feedback you're giving me and I'm doing my best to investigate/test as well. Thank you very much!

dvernier, 2016-09-09:
Hello again,

I think I now understand that you are using our sensors with an Arduino,
our shield, and LabVIEW on a computer. This is an very  unusual,
complicated system. I have never done it. There are so many things that
could be wrong, it is difficult to even troubleshoot. Why are you using
such a complex system? Can you call me and let's discuss your goals here
and how you could best get the job done. Call 1-888-837-6437.The only catch
is I have to be out of the offfice from 8:30 am Pacific time until about 1
pm Pacific time.

>>I also have one more question but for the pH sensor this time. When I
measured the pH (already calibrated using the buffer solutions) of my
280-gallon fish tank, the final reading fluctates from 9-10. However, when
I immediately move the sensor to a cup/glass of the same aqarium water, the
final reading is around 7.2-7.4. This test was done with no other sensors
in the water at the time. Does the volume of water have an effect on the
output of the pH sensor?

This is not reasonable. It implies to me that there is some electrical
signal (from a pump or some other sensor) getting into you water. Try the
same test with no electrical device of any kind in or touching the water,
except our pH probe.

Dave Verniier

David L. Vernier
Founder and Co-President
Vernier Software & Technology
13979 SW Millikan Way
Beaverton, OR  97005
503-277-2299
1-888-Vernier (1-888-837-6437)
dvernier@vernier.com
www.vernier.com

YukiSakuma, 2016-09-16:
Greetings Sir Vernier! I apologize for a late reply, thank you so much for your feedback. I was working with the sensors and the information you gave me. These are my findings:

- The pH and Stainless Steel Temperature Sensors work fine in giving real-time measurements over extended periods of time

- My current DO probe interferes with the other sensors so I have no choice but to measure dissolved oxygen via sampling in an electrically isolated system.

- After keeping them for a few days (according to their instructions) both the DO Probe and Ammonium ISE produce flat 0V for both the raw and converted voltage when printed in the appropriate Arduino COM Port Monitor. I have never had this problem with the other two sensors.

The only interface I'm using is a Sparkfun Vernier Sensor Arduino shield coupled with a Red Sparkfun Arduino Uno board and the official Arduino software. No lab-quest or related Vernier software/hardware. The Arduino is solely powered by the USB connection to my laptop.

This leads me to my next question, what does it mean when both my Ammonium ISE and DO register flat 0V readings in the Arduino COM Monitor? I limited them only to sampling purposes and stowed them when they were not in use.

Thank you again for helping me.

Smythics, 2016-09-16:
Hi. Dave's not in the office today so I'll try to help.

I went and talked to one of our biologists and he stressed that these sensors are not intended to be used _in situ_ for long periods of time, but rather calibrated and then used to measure a sample. When you need another reading you need to calibrate again (or take your chances...) and measure.  The issue is that the membranes for both of these sensors change over time.

I'll refer you back to Dave's comment that our sensors may not be the best for this application. We've had some inquiries about this for tilapia projects in their classrooms, etc. but I haven't heard of any successful "hands free" monitoring systems.

If I've misinterpreted your set up, please let me know. A photo of the set up may be helpful.

Tom

dvernier, 2016-09-17:
Hello,

I am away on vacation until Oct 3rd. There are other people at the office who can help you, but to know who to recommend, I am still trying to figure out what exactly you are doing.  We have never used Arduino
With LabVIEW. If that is what you are doing, contact Sam Swartley, sswartley@vernier.com.

The fact that you are reading 0 volts, makes me suspicious that you are not reading the voltage properly.

Here is something to try, which could help pin this down:  with the program running, swap a working sensor with a non-working sensor. Do you still get a 0 reading? Of course, the reading cannot be right, but it should at least be non-zero.

As I said last time, the fastest way to resolve this would be for you to call our office.

Dave Vernier

Sent from my iPhone

> On Sep 16, 2016, at 3:31 PM, YukiSakuma <notifications@github.com> wrote:
> 
> This leads me to my next question, what does it mean when both my Ammonium ISE and DO register flat 0V readings in the Arduino COM Monitor? I limited them only to sampling purposes and stowed them when they were not in use.