VernierST/arduino #4: quadratic calibration curves sensors

Basazababo, 2016-12-23:
Hey,
We bought ETH-BTA from you and am trying to use it with arduino uno and i found that You have a bunch of arduino codes for most of your products;
do you by chance have a sample codes for the sensor with quadratic calibration curves? @davidlim

Smythics, 2016-12-23:
I put this together based on the basic Vernier Analog Sensor sketch. The default calibration for this sensor is to take the voltage to the -2.995 power and multiply that by the coefficient (0.9054). 

[VernierEthanolSensor.zip](https://github.com/VernierSoftwareTechnology/arduino/files/671100/VernierEthanolSensor.zip)

There is additional information in the sensor user manual regarding calibration.

dvernier, 2016-12-23:
I think that this sensor is handled in the VernierAnalogSensorAutoIDDisplay sketch, but I have not tested it much.

closed 2016-12-23 by dvernier

Basazababo, 2016-12-24:
@dvernier, @Smythics; Thank you for your prompt response;
i will try this