VernierST/arduino #8: Calibrate for sensor NH4-BTA

kakaducsy, 2017-07-28:
Im bought sensor NH4 ( https://www.vernier.com/products/sensors/ion-selective-electrodes/nh4-bta/ )
I want use arduino calib with NH4 
this it document NH4 ( https://www.vernier.com/files/manuals/nh4-bta.pdf )
E = Eo + m(ln a) 
where E is the measured voltage, Eo is the standard potential for the combination of
the two half cells, m is the slope, ln is the natural logarithm, and a is the activity of
the measured ion species. 
I dont understand Eo,How do I determind Eo ?

Smythics, 2017-07-28:
Eo is the response from the sensor in mV. Here is a bit more background:

Assuming the ionic strength is fairly constant, the Nernst equation may be rewritten
to describe the electrode response to the concentration, C, of the measured ion
species:
E = Eo + m(ln C) 

The sensor converts a 0-5 V reading into a mV output. The mV reading then corresponds to the concentration of the ion for the particular ISE you are using.

If you are interested in using an ISE with an Arduino, you need to create a sketch that performs the following conversions:
1) Convert raw Arduino reading to a 0-5 V range
2) Convert 0-5 V range to mV using E = 137.55*Eo -0.1682 where E is the response in mV and the Eo is the voltage from the sensor.
3) Convert mV to an ISE sensor reading

I will note that we do not recommend this combination due to the nature of the sensor and complexity of continuous need for calibration. We have posted an Arduino sketch on github that reads a Vernier ISE sensor. This sketch can be found at:
https://github.com/VernierSoftwareTechnology/arduino/blob/master/VernierBNC/VernierBNC.ino

kakaducsy, 2017-07-29:
thank you very much :)

Smythics, 2017-09-06:
Sorry for the delayed response. Yes - all of the ISE's would need the same treatment. Please note that using an ISE with Arduino is not a recommended solution.

closed 2017-09-06 by Smythics