VernierST/arduino #10: Vernier ISE for Ammonium, Nitrate and Potassium not working

preity1802, 2020-08-13:
The library(VernierLib) provided by Vernier does not work with the ISEs I bought from Vernier. It gives 0.00 output for all the ISEs.
Also, when I capture the value from Analog2 port, the values are not stable for High or low-value solutions. Please help!

The attached screenshot is for the type of output when I am reading it from Analog2. It is averaged for 10 consecutive values. It looks the same for both 1 PPM and 100 PPM solutions.

Also, one thing I noticed that for Ammonium ISE, the ISE behaves as -ve linear way, which is just opposite of what is mentioned on the website. Similarly, for nitrate, it behaves in a +ve linearly way but opposite to what is mentioned on the website(-ve linearly).

13-08-2020 17:50 PM: I am attaching the four files having values for 1 PPM and 100 PPM from Nitrate ISE with a simple analog read the program and an analog read program with averaged value.
Data is almost similar for both 1 PPM and 100 PPM.

Program with averaging:
void setup() {
Serial.begin(9600);
Serial.println("CLEARDATA"); //clears up any data left from previous projects

Serial.println("LABEL,Time,Timer,sensorVoltage"); //always write LABEL, so excel knows the next things will be the names of the columns (instead of Acolumn you could write Time for instance)

Serial.println("RESETTIMER"); //resets timer to 0
}

void loop() {
float sensorVoltage,sensorReading; // declares a variable named sensorVoltage
int numberAveraged = 10; //number of readings averaged for reading reported
int i,count;
int sum = 0;
float voltage;

for (i = 0 ; i < numberAveraged; i++)
{
count = analogRead(A2); //read 0 to 5 volt analog lines Analog 2
sum = sum + count;
}
sensorVoltage = ((sum / numberAveraged) * 5.0)/ 1024.0; //convert average count to voltage (0 to 5 volt input)
Serial.print("DATA,TIME,TIMER,sensorVoltage");
Serial.println(sensorVoltage);
delay(100); // delay in between reads in milliseconds
}

Program without averaging:
13-08-2020_Nitrate-1 PPM_ averaged.xlsx
13-08-2020_Nitrate-1 PPM_without averaged.xlsx
13-08-2020_Nitrate-100 PPM_ averaged.xlsx
13-08-2020_Nitrate-100 PPM_without averaged.xlsx
void setup() {
Serial.begin(9600);
Serial.println("CLEARDATA"); //clears up any data left from previous projects

Serial.println("LABEL,Time,Timer,sensorVoltage"); //always write LABEL, so excel knows the next things will be the names of the columns (instead of Acolumn you could write Time for instance)

Serial.println("RESETTIMER"); //resets timer to 0
}

void loop() {
float sensorVoltage,sensorReading; // declares a variable named sensorVoltage
int sensorValue = analogRead(A2);
sensorVoltage = sensorValue * 5.0 / 1023;// Converts the count to the sensor voltage
Serial.print("DATA,TIME,TIMER,sensorVoltage");
Serial.println(sensorVoltage);
delay(100); // delay in between reads in milliseconds
}
[13-08-2020_Nitrate-1 PPM_ averaged.xlsx](https://github.com/VernierST/arduino/files/5069045/13-08-2020_Nitrate-1.PPM_.averaged.xlsx)
[13-08-2020_Nitrate-1 PPM_without averaged.xlsx](https://github.com/VernierST/arduino/files/5069046/13-08-2020_Nitrate-1.PPM_without.averaged.xlsx)
[13-08-2020_Nitrate-100 PPM_ averaged.xlsx](https://github.com/VernierST/arduino/files/5069047/13-08-2020_Nitrate-100.PPM_.averaged.xlsx)
[13-08-2020_Nitrate-100 PPM_without averaged.xlsx](https://github.com/VernierST/arduino/files/5069048/13-08-2020_Nitrate-100.PPM_without.averaged.xlsx)
[image]

dvernier, 2020-08-13:
Hello,

I will try to answer your questions below.

>>The library(VernierLib) provided by Vernier does not work with the ISEs I
bought from Vernier. It gives 0.00 output for all the ISEs.

How are you using the ISE’s?. Are you attaching them to the Vernier Shield?
What Arduino are you using?

>>Also, when I capture the value from Analog2 port, the values are not
stable for High or low-value solutions. Please help!

The library is only for use on the Analog 1 port of the shield. You can
write your own programs to read the analog voltage from Analog 2, if you
want.

>>Also, one thing I noticed that for Ammonium ISE, the ISE behaves as -ve
linear way, which is just opposite of what is mentioned on the website.
Similarly, for nitrate, it behaves in a +ve linearly way but opposite to
what is mentioned on the website(-ve linearly).

I am not clear what you mean here. Certainly the calibration of the ISEs is
very non-linear. Also, for the ISE’s the library will read mV, not
concentration.

>>I am attaching the four files having values for 1 PPM and 100 PPM from
Nitrate ISE with a simple analog read the program and an analog read
program with averaged value.

Data is almost similar for both 1 PPM and 100 PPM.

I am not clear on what you are doing here. It seems that you are writing
your own program, not using the VernierLib library, which is fine, but I
assume you are reading raw voltage. That should change with concentration.

What educational institution are you with. I assume this is a project to
teach students coding or doing student projects. That is what the library
was designed for.

Dave

David L. Vernier
Founder and Co-President
Vernier Software & Technology
13979 SW Millikan Way
Beaverton, OR  97005
503-277-2299
1-888-Vernier (1-888-837-6437)
dvernier@vernier.com
www.vernier.com

preity1802, 2020-08-13:
Hi, thank you for your reply. I could not use vernier library as it was not giving any output with analog1 port. So, I used my code to read from analog 2. My main concern is that the outputs doesn’t show any difference between the 1 mg/l soln and 100 mg/l solution, voltage should have some difference as per the ISE manual attached on vernier website. I used vernier with 2 bta and 2 btd ports board and arduino uno.

preity1802, 2020-08-13:
#vernier board compatible
With arduino uno

dvernier, 2020-08-13:
OK. I have some other questions:

Are you using the Vernier Arduino Shield?
 https://www.vernier.com/product/vernier-arduino-interface-shield/
<https://www.vernier.com/product/vernier-arduino-interface-shield/>

Have you tried the ISE with one of our lab interfaces?
https://www.vernier.com/downloads/labview-virtual-instrument-downloads/labquest-interfaces/

On your do-it-yourself programs, are you reading the 0 to 5 volt line? See:
https://www.vernier.com/engineering/arduino/arduino-online-guide/connecting-vernier-sensors-to-arduino/

Yes the voltage should vary as you change the concentration.

Have you looked at the manual for the ISE that you bought. All manuals are
on our web site.

Dave

David L. Vernier
Founder and Co-President
Vernier Software & Technology
13979 SW Millikan Way
Beaverton, OR  97005
503-277-2299
1-888-Vernier (1-888-837-6437)
dvernier@vernier.com
www.vernier.com

preity1802, 2020-08-13:
Yes, I am using this vernier shield. And I have been through the lab manuals. I could not check with any of the lab interfaces.

preity1802, 2020-08-13:
I used Vernier Arduino interface shield , where no wiring is required . I just inserted the ISE cable into the shield’s Analog 2 port.

dvernier, 2020-08-13:
OK. If you are using the Analog 2 port (I wonder why?) then you have to write your own program to read the A3 input line. A1 and A2 are connected to Analog 1 and A3 and A4 are connected to Analog 2. A3 is the 0 to 5 volt input used by our ISE sensors.

Dave

preity1802, 2020-08-14:
https://www.vernier.com/til/3909
With reference to the above link,

**Analog 1
Sensor pin 1 (+/-10 V output line) – Arduino pin A1
Sensor pin 2 (Ground) – Arduino ground
Sensor pin 3 (Vres/resistance reference) – Arduino pin A4
(also for I2C autoID)
Sensor pin 4 (AutoID) – Arduino pin A5 (also for I2C autoID)
Sensor pin 5 (+5 Volt power) – Ardino 5V
Sensor pin 6 (0-5 V output line) – Arduino pin A0

Analog 2
Sensor pin 1 (+/-10 V output line) – Arduino pin A3
Sensor pin 2 (Ground) – Arduino ground
Sensor pin 3 (Vres/resistance reference and I2C input line- Arduino pin A4)
Sensor pin 4 (AutoID) and I2C output line– Arduino pin A5)
Sensor pin 5 (+5 Volt power) – Ardino 5V
Sensor pin 6 (0-5 V output line) – Arduino pin A2**

It says A2 is 0-5 V output line for Analog2

preity1802, 2020-08-14:
Reading analog pin A1 gives a constant voltage output of 3.12 Volt when the ISE is connected to Analog1 in the interface shield for both 1 PPM and 100 PPM concentration.

[14-08-2020_Nitrate-1 PPM_ averaged_A1.xlsx](https://github.com/VernierST/arduino/files/5072802/14-08-2020_Nitrate-1.PPM_.averaged_A1.xlsx)
[14-08-2020_Nitrate-1PPM_without averaged_A1.xlsx](https://github.com/VernierST/arduino/files/5072803/14-08-2020_Nitrate-1PPM_without.averaged_A1.xlsx)
[14-08-2020_Nitrate-100 PPM_ averaged_A1.xlsx](https://github.com/VernierST/arduino/files/5072804/14-08-2020_Nitrate-100.PPM_.averaged_A1.xlsx)
[14-08-2020_Nitrate-100 PPM_without averaged_A1.xlsx](https://github.com/VernierST/arduino/files/5072805/14-08-2020_Nitrate-100.PPM_without.averaged_A1.xlsx)

preity1802, 2020-08-14:
Reading analog pin A0 and A2 give voltage output of very little difference for both 1 PPM and 100 PPM concentration when the ISE is connected to Analog1 in the interface shield. I have attached few sample outputs recorded .
[14-08-2020_Nitrate-1 PPM_ averaged_A2.xlsx](https://github.com/VernierST/arduino/files/5072856/14-08-2020_Nitrate-1.PPM_.averaged_A2.xlsx)
[14-08-2020_Nitrate-1PPM_without averaged_A0.xlsx](https://github.com/VernierST/arduino/files/5072857/14-08-2020_Nitrate-1PPM_without.averaged_A0.xlsx)
[14-08-2020_Nitrate-1PPM_without averaged_A2.xlsx](https://github.com/VernierST/arduino/files/5072858/14-08-2020_Nitrate-1PPM_without.averaged_A2.xlsx)
[14-08-2020_Nitrate-100 PPM_without averaged_A0.xlsx](https://github.com/VernierST/arduino/files/5072859/14-08-2020_Nitrate-100.PPM_without.averaged_A0.xlsx)
[14-08-2020_Nitrate-100 PPM_without averaged_A2.xlsx](https://github.com/VernierST/arduino/files/5072860/14-08-2020_Nitrate-100.PPM_without.averaged_A2.xlsx)
[14-08-2020_Nitrate-1 PPM_ averaged_A0.xlsx](https://github.com/VernierST/arduino/files/5072855/14-08-2020_Nitrate-1.PPM_.averaged_A0.xlsx)

preity1802, 2020-08-14:
[image]
 For reference:
https://learn.sparkfun.com/tutorials/vernier-shield-hookup-guide?_ga=1.220467931.257332638.1417470376#vernier-shield-pin-out-and-configuration

preity1802, 2020-08-14:
[image]

preity1802, 2020-08-14:
///OK. If you are using the Analog 2 port (I wonder why?) then you have to write your own program to read the A3 input line. A1 and A2 are connected to Analog 1 and A3 and A4 are connected to Analog 2. A3 is the 0 to 5-volt input used by our ISE sensors.//

Reading A3 gives a constant output of 3.11 V when the  ISE is connected at analog 2 of the shield.

dvernier, 2020-08-14:
Please let me know:

- do you have any other Vernier BTA sensors to try to compare results?
- do you have a Vernier lab interface like a LabQuest or a Go!Link so that
you can try the ISE there?
- have you tried the ISE in Analog 1, just as a test, either with the
VernierLib library or without?
- have you tried feeding a know voltage signal into the A3 line to see if
you can read it properly?
- What educational institution are you with?

Dave
David L. Vernier
Founder and Co-President
Vernier Software & Technology
13979 SW Millikan Way
Beaverton, OR  97005
503-277-2299
1-888-Vernier (1-888-837-6437)
dvernier@vernier.com
www.vernier.com

preity1802, 2020-08-14:
I have ISEs only from vernier

No Lab interface

Have tried ISE in analog 1, please refer comments Above. I have already attached data from analog 1.

Vernierlib dont give any value. It just outputs 0.0 . I have already stated this in my issue description above.

I am associated with a research laboratory in India. So, ISEs with microcontroller are my prime concern.

dvernier, 2020-08-14:
You said:
>>I have ISEs only from vernier
Do you have more than one?

and I ask again:
 have you tried feeding a known voltage signal into the A3 line to see if
you can read it properly? I want to make sure of two things:
   is your Arduino and the shield working properly to read a voltage, and
it your sketch working? The voltage should be applied between the two pins
closest to the tab on the BTA connector.

Dave
David L. Vernier
Founder and Co-President
Vernier Software & Technology
13979 SW Millikan Way
Beaverton, OR  97005
503-277-2299
1-888-Vernier (1-888-837-6437)
dvernier@vernier.com
www.vernier.com

preity1802, 2020-08-14:
Can you please tell me from where should I read A3? Analog 1 or Analog2? Because the specification given on your website say “ Analog 2
Sensor pin 1 (+/-10 V output line) – Arduino pin A3 “. So, am I supposed to read +/-10 V output line? 
Also, only Nitrate Ise I have tried with arduino on Analog 2 port and it give constant output which I have already mentioned in the above comments

dvernier, 2020-08-14:
Each of the two analog connectors on the Vernier interface shield has two
analog input lines. I made a mistake in a couple of my emails on the line
numbers. It is the classic mixup of counting starting with 0. Below is
correct:

Analog 1 uses the Arduino A0 and A1 input lines.
Analog 2 uses the Arduino A2 and A3 input lines.

So if you want to test the Arduino, the shield, and your code:
one shield connector labelled "Analog 1" connect the two pins closest to
the tab to a small known voltage and try to read it as A0.
one shield connector labelled "Analog 2" connect the two pins closest to
the tab to a small known voltage and try to read it as A2

Dave
David L. Vernier
Founder and Co-President
Vernier Software & Technology
13979 SW Millikan Way
Beaverton, OR  97005
503-277-2299
1-888-Vernier (1-888-837-6437)
dvernier@vernier.com
www.vernier.com

preity1802, 2020-08-17:
It works well with reading A0 using the Analog1 port.
The issue has been resolved.
Thank you for your kind help.

closed 2020-08-17 by preity1802